WendyTeslaburger/WT-312 #6: Schematic mismatch

maju101, 2021-05-16:
I seems that the pdf schematic and the PCB do not match. The schematic show 2 opamps (U1A and U1B), however the PCB only has room for 1.

I would greatly appreciate it if you could update the pdf schematic (and perhaps include a partslist). If you could do this update I will try to build it and add a partslist via Pull Request if I get it to work.

Thanks

WendyTeslaburger, 2021-05-16:
Hello, Thank you for your interest in my project!

U1A and U1B are both the same opamp. The LMC6482 is a dual opamp ([datasheet](https://www.ti.com/lit/ds/symlink/lmc6482.pdf?ts=1621152332898))
Splitting a component into multiple schematic subparts is a very common design technique.

However, after experimenting a bit with this module there are some slight variations I changed some component values for better performance. I should update these values into a new schematic PDF.

Best regards,
WT

maju101, 2021-05-16:
Ah ok cool.
Can I ask a couple more questions? 

- The L1 coil is marked as z60. I'm assuming that's an impedance? If so, how does this translate to a coil rating in Henry?
- What size SMD components are you using? (i.e. 1206, 0603 ...)

Thanks
Cj

(and as promised, once I have everything assembled I'll add the full partslist via pullrequest so others who might be interested in this project can just buy everything without bothering you ;) )

WendyTeslaburger, 2021-05-18:
L1 is actually a filter bead. I could change the refdes to something like "FB1" to avoid confusion.
Z60 means that it has a 60ohm impedance at 100MHz. The filter bead is used to filter out protentional noise to the opamp. The noise could for example originate from a DCDC circuit to generate the 9v.

It is good practice to have it there, but I think the risk is low that the circuit would behave erratic if you just put a 0 ohm resistor there, or short the terminal with a solder blob.

The component size for the SMD components is 0603. For me this is the smallest size that is still comfortable to hand-solder.

jijzult, 2021-09-24:
I'd love a complete/matching BOM/schema too :) I'm following with great interests. I'll be trying to then make one with a LPR6235 coilcraft, so get a an ultrasmall size..

WendyTeslaburger, 2021-09-24:
I will try to get a BOM .csv file on the repo this weekend. Sorry for all the delays, I wish I had more time to work on this project.
Meanwhile some other kind strangers are testing out the design for me.

Regarding the LPR6235: I have doubts if this will work.
The LPR6235 is a tiny inductor and does not have a lot of material. This means that it will saturate quite fast.
The LPR6235-253LMR for example already saturates at 1.3A of primary current. This will not be sufficient for stronger e-stim output pulses.

maju101, 2021-09-24:
I was going to provide the bom as promised once I built it so I am sure
it's correct. But Mouser has absolutely massive delays. Currently expecting
my parts at the start of January...

closed 2023-05-21 by WendyTeslaburger