WendyTeslaburger/WT-312 #7: small batch production run

McSwitchy, 2021-07-14:
Hi WendyTeslaburger - I am super impressed w/ your project and have been hoping someone would put together a small production run of the PCBs. 

1. has anyone produced a batch of assembled PCBs or is planning to?

2. if not, I guess I would like to give it a shot. I've gotten pcb fab + assembly quotes from 1) circuithub.com and 2) pcbway.com in the past. However:

circuithub requires CAD files and not just gerbers - would you be willing to add those to this repo?
> For Altium projects, upload your.pcbdoc and all the relevant.schdoc files
> For Eagle projects, upload your.brd file and one.sch file
> For KiCAD projects, upload your.kicad_pcb, -cache.lib and.sch
> - http://docs.circuithub.com/en/articles/62996-can-i-upload-and-quote-gerber-files
> BOM should be in RFC4180
> - http://docs.circuithub.com/en/articles/1968003-can-i-upload-my-bom-to-my-project

PCBway can work from gerber files, but needs a **centroid** file for each component for their pick-n-place process. See their requirements here: https://www.pcbway.com/helpcenter/pcb_assembly_ordering/

**first of all, is it ok w/ you for me to do a production run of this project? I would probably make 10-20 depending on price breaks and sell them at cost to anyone interested - alternatively I could just make a couple for personal use.** please let me know what you would prefer - I want to respect your wishes.

That said, assuming you support a small production run... Would you be open to: 

1. creating the centroid file for PCBWay, or
2. providing the original atium design files so I can use circuithub;
2. and helping me pick good parts for the BOM
3. and providing advice (if any) on the optimal pcb and assembly options offered by pcbway (i.e. extra-thick copper layer for antenna / power / emf shielding etc)? See pcbway fab options in screenshot below

if not, I guess I would recreate the design in kicad or eagle and try out circuithub.

THanks! Other people interested in getting a board, chime in please so I can estimate interest.

McS

---

### PCBWAY manufacturing parameters

[image]

[image]

WendyTeslaburger, 2021-07-15:
Hey,
Thanks for your interest in the project! I will try to answer all your questions.

I have made a small batch of 10 PCBs at JLCPCB for testing and debugging the design. I hand-soldered them myself.
But as far as I'm aware, there has been no other production run.

I'm no copyright lawyer, but I tried to keep these project as open as possible so I included the Creative Commons Zero v1.0 licence. This should allow anyone to do whatever they want with the design. So I am 100% down with you running some production on these modules.

I use Altium Designer for my design work.
For this project I did not yet include the source documents because I was not sure if any personal information was inside .SchDoc or .PcbDoc metadata. For the WT-108 project I did some research and felt comfortable enough to commit the source documents. I will make a commit with the Altium source documents so you can use them (Circuithub) or even contribute to the project if you have an Altium licence. 

Regarding the actual design itself: I had the chance to experiment with these boards and found some smal-ish issues. If any serious production run would be done, it would be best to make some changes to the design before making the modules.
Some are simple like substituting some component values with different values. But others would benefit from a small PCB redesign.

If you like, I could make these changes over the weekend and make a commit. Then we can have a look together at manufacturing options. 

Best regards,
WT

McSwitchy, 2021-07-15:
Hi,

Most excellent! When you get the time to make the minor tweaks and then commit the updated atium files etc to the repo, I'll dig in and start trying to source parts and setting up a BOM. I'll try to do some of the heavy lifting there and also by soliciting a set of RFQs from the fab places. Then we can refine the BOM, order quantity, etc.

I'm glad I reached out, I've been meaning too for a year now :)

rexorient, 2021-07-22:
I would be interested in buying some assembled boards.

WendyTeslaburger, 2021-07-23:
Hello,
Sorry for the delay.
I have already made the changes to the schematic and committed the schematic in PDF together with the source .SchDoc file.
Next up I will change the routing a bit on the module PCB and commit that together with the Gerbers.

Regarding the BOM: I think the schematic is now as good as it gets with all the open issues fixed.
I will export a BOM list from Altium still today, then we can have a look if all components are available or if they need to be swapped for a different supplier (the component marking is crazy right now!)

Best regards,
WT

Deor, 2021-07-28:
Hi All,
I've just found this project and its just what i was looking for!
As I had also just put in an order with JLCPCB for some MK-312BT boards i decided to add this to the order too.
I know its still WIP, but for another £1.44 i thought why not!
If nothing else it will be good SMT soldering practice :)

KnkstrKex, 2021-08-05:
I'm also interesting in buying some (e.g. 5) completed boards.

h2omatt, 2022-09-11:
Did anyone make a few of these I am interested in buying a few to test with

EmiliaThorsen, 2024-06-08:
Any possibility for centroid files to be generated and added to the repo for production at jlcpcb/pcbway? I do not have altium so i can't generate the files myself, attempting to import the save files into kicad seems to have gotten the placements correctly, but i do not fully trust that it did everything correctly.

Otherwise a full conversion of the project to kicad would be amazing, but that is a lot to ask for so no need to, its only the centroid files for the pick and place i need to have it produced.