WendyTeslaburger/WT-312 #8: Is there any software other besides the example?

Misaka13514, 2022-01-10:
Hi WendyTeslaburger! I am very interested in WT-312 because of its small size, low cost, etc.

I have completed the V2.0 of the hardware production.
[image]

When I tested the ESP32 software example, I found that neither V1 nor V2 software seemed to work properly on my device. It always enters the errorloop. (Sorry I don’t know how to view debugging information. I will try my best if you teach me.)

btw, is there any software other besides the example? Can the wireless function of esp32 be used to control pulse?

86lotus, 2023-03-31:
Hi .. did you ever get this working? I'm about to embark on a build and would love to know if yours worked out.

Do you have a BOM? Doesn't seem to be one in the project ...

Misaka13514, 2024-03-26:
@86lotus Sorry for the late reply. Yes I got this working recently. There is a BOM at https://github.com/WendyTeslaburger/WT-312/blob/master/Hardware/WT312-Mouser-BOM.xls